mdfbaam/ORNL-Slicer-2-Issue-Tracker #29: Update icon to reflect Slicer "2"

ericblack04, 2021-03-16:
## Expected Behavior
Self-Explanatory.

## Actual Behavior
Self-Explanatory

## Possible Solution
Maybe just put a nice "2" or "II" in the icon so I can tell them apart while I have both installed.

## Steps to Reproduce the Problem
Self-Explanatory.  Ideally, bullet or numbered list.

## Specifications
Platform, machine specs, anything else you think might be relevant.

ericblack04, 2021-03-16:
I can't flair it with labels but Alex if you see this I think it's understood that this is low priority wishlist material!

aroschli, 2021-03-16:
> I can't flair it with labels but Alex if you see this I think it's understood that this is low priority wishlist material!

closed 2021-03-31 by mborish