mdfbaam/ORNL-Slicer-2-Issue-Tracker #39: "Total Time Estimate" value incorrect

ericblack04, 2021-04-29:
## Expected Behavior
Sliced a print, expected a valid number for expected time.

## Actual Behavior
Estimated time was -8 sec.

## Possible Solution
Self-Explanatory

## Steps to Reproduce the Problem
Force minimum layer time was enabled in the cooling section using modify feedrate set to 180s.

## Specifications
Platform, machine specs, anything else you think might be relevant.

ericblack04, 2021-04-29:
Bug, core-gcode, low

ericblack04, 2021-04-29:
I re-sliced another STL with different settings and it seemed to estimate time just fine, still have the modify feedrate set to 180s.

ericblack04, 2021-04-29:
Sliced a simple box STL and it estimated time just fine. Switched out the STL without changing any other settings and it gave me -8 sec again for the estimated time.

aroschli, 2021-04-29:
@ericblack04 can you provide a project file where it does this? The project system is fixed on our version so I should be able to recreate your issue.