mdfbaam/ORNL-Slicer-2-Issue-Tracker #42: Copy and paste of STLs not functioning properly

ericblack04, 2021-04-30:
## Expected Behavior
Copy existing STL, paste. Should duplicate the first STL and allow for independent control of both.

## Actual Behavior
Copied and pasted STL. Only 1 STL was visible in the build volume but the object list showed 2 identical file names. Tried copy and paste several more times with similar results. Then attempted to delete the STLs which caused a crash to desktop.

## Possible Solution
Self-Explanatory

## Steps to Reproduce the Problem
Self-Explanatory.  Ideally, bullet or numbered list.

## Specifications
Platform, machine specs, anything else you think might be relevant.

ericblack04, 2021-04-30:
Bug, core-geometry, medium

aroschli, 2021-04-30:
Deleted because issue already listed in #11

closed 2021-04-30 by aroschli