mdfbaam/ORNL-Slicer-2-Issue-Tracker #44: Slicing Multiple STLs Bug

ericblack04, 2021-04-30:
## Expected Behavior
Slicer should slice multiple STLs without issue. Travel between two STLs should be:

Travel Lift Z
XY Travel
Travel Lower Z

## Actual Behavior
Extruder is moving to a different Z height within a layer when it moves to a different STL.

For example, my first layer is Z-4.5. The first Z travel and extrusion moves all take place at Z-4.5. However, when it goes to travel to the next STL, instead of it doing an expected lift, travel, lower, it does a lift, XYZ travel, then a lower.

## Possible Solution
Self-Explanatory

## Steps to Reproduce the Problem
Self-Explanatory.  Ideally, bullet or numbered list.

## Specifications
Platform, machine specs, anything else you think might be relevant.

ericblack04, 2021-04-30:
Bug, core-gcode, high

mborish, 2021-05-03:
Fix will appear in v.94 release.

closed 2021-05-03 by mborish