mdfbaam/ORNL-Slicer-2-Issue-Tracker #48: Expand G-code layer viewing options

ericblack04, 2021-07-08:
## Expected Behavior
In Slicer 1, we had the ability to select between "Show Complete Code" "Show Layer Range" and "Show Single Layer" as well as two slider bars to visualize g-code in 3D. In addition we could "Hide Travel" and "Hide Support" and several others.

## Actual Behavior
Currently in Slicer 2, only having 2 layer number boxes with "Hide Travel" and "Hide Support". These boxes feel very clunky to either scroll in to visualize the g-code or to type your desired layer in. The sliders from Slicer 1 were a great way to quickly visualize where in your overall print you were at as well as jump between locations quickly. It felt like a very natural way to visualize layer ranges as well.

## Possible Solution
Expand G-code layer viewing options (ideally at least up to the level Slicer 1)

ericblack04, 2021-07-08:
Core-ui Mid-Priority Wishlist

aroschli, 2021-10-15:
Goes along with #61