mdfbaam/ORNL-Slicer-2-Issue-Tracker #53: Maintain STL Coordinates

aroschli, 2021-09-28:
## Expected Behavior
Self-Explanatory.

## Actual Behavior
Loaded STL files are automatically centered in the build volume. There is no way to disable this functionality, making it difficult to properly align multiple files.

## Possible Solution
A setting in the preferences to disable automatic centering of STL files.

## Steps to Reproduce the Problem
Self-Explanatory. 

## Specifications
A loaded STL file is automatically centered, essentially ignoring the coordinate system used when saving the STL. We need the ability to maintain the coordinate system, not automatically rotate/translate the object, so the user can apply the same transformation to multiple files for proper alignment.

AESjay, 2022-01-12:
@aroschli just checking in on this issue to see if there's been any movement? I think this is the same as issue #58

aroschli, 2022-01-12:
@AESjay thanks for checking in on this. We have not made progress on this issue yet. It is very similar to #58, but it is not quite the same. This one is specific to loading a part and maintaining the part coordinate system, which could mean loading a part and have it be nowhere near the build volume. #58 refers to the automatic shifting to avoid collisions, so still centering the part but ensuring it doesn't sit atop another part.

closed 2022-01-26 by mborish