mdfbaam/ORNL-Slicer-2-Issue-Tracker #94: Scale, Rotate, Move values don't appear when in dark mode

d-portero, 2022-07-13:
## Expected Behavior
The values for the part size, current rotation, scaling appear when the button is clicked

## Actual Behavior
Blank spaces appear 

## Possible Solution
Self-Explanatory

## Steps to Reproduce the Problem
-Change to dark mode
-upload stl
-attempt to scale/rotate/move object

## Specifications
Beta Version 0.99.3
[image]
[image]
[image]

its-a-bee, 2022-08-16:
@d-portero, sorry for the late response. Can you tell me more about your computer/operating system? The UI elements you have identified display correctly on my machine, as shown below. 

[image]

I'm not sure what could cause the discrepancy.

d-portero, 2022-08-17:
The computer is a Lenovo laptop running windows 10. Intel i7 processor. Is there anything else you would like to know?

its-a-bee, 2022-08-24:
We have been investigating on our end and it seems to be an issue with the installer. We will work to get this fixed asap. Thanks for reporting the issue.

its-a-bee, 2022-11-29:
I believe this problem only occurred during sessions in which the theme was switched from Light to Dark. There was an issue when updating the theme mid-session that caused the specified submenus to not be colored correctly; it updated the text color to white, but did not update the background, causing the text to "disappear". When updating from Dark to Light, the Dark theme colors remained, which did not make the values disappear but did look a bit odd. 

For now -- if Slicer is exited in dark mode and then restarted, the menus and values should appear correctly. 

I have fixed the theme updating issue and in the next release this should be resolved.

closed 2023-01-25 by aroschli