mdfbaam/ORNL-Slicer-2-Issue-Tracker #126: Visualization Slider can go negative

AESjay, 2023-06-22:
## Expected Behavior
"First Segment" and "Last Segment" box should stop at zero.

## Actual Behavior
Sometimes when sliding the bars to the left with the mouse, the boxes will display a "-1" segment, and then if you slide the bottom bar back to the right, the visualization does not come back.  You have adjust a different slider to get it back to zero or positive before it will display any preview.  

## Possible Solution
Check for negative numbers and stop at zero.

## Steps to Reproduce the Problem
Self-Explanatory.  

[image]

AESjay, 2023-06-22:
Low Priority, Visualization, Bug fix

aroschli, 2023-12-22:
Moved to new tracker

closed 2023-12-22 by aroschli