mdfbaam/ORNL-Slicer-2-Issue-Tracker #128: Feature Request - Lock STL Positions

AESjay, 2023-06-22:
## Observed:
Since it is often necessary to highlight an STL to change settings, it is all too easy and common to accidentally rotate the STL that is highlighted when trying to rotate the viewport.

## Possible Solution
Incorporate a check box (maybe by each STL?) that would lock it's position, rotation, and scale and would not allow any accidental movements due to mouse click and drags when trying to rotate the view. 
[image]

AESjay, 2023-10-19:
@aroschli @mborish - Bringing this back up as this has really become a huge issue. The number of times we have a model selected and are trying to simply translate it but make one errant mouse click and the whole model just spins at some random location, means we have to completely delete and reload the STLs and start over with any settings (undo doesn't work for model rotations).  

Possible to implement a lock feature or change this behavior?

aroschli, 2023-12-22:
Moved to new tracker

closed 2023-12-22 by aroschli